ohAnd/EOS_connect #2: Quelle für Lastprofil ("gesamtlast") um influxdb erweitern bzw. Wallbox Anteil herausrechnen

WolfImBusch, 2025-03-24:
Hi @ohAnd ,

als Quelle für das Lastprofil kann ich an Home Assistant andocken. Das ist schon ziemlich cool!

Allerdings wird das Lastprofil insofern verfälscht, als dass dort auch das Laden des E-Autos mit enthalten ist. Wäre es möglich die config.yaml um das Lastprofil der Wallbox zu erweitern, die dann von der Gesamtlast subtrahiert wird? 

Prima wäre auch, wenn EOS_connect eine Unterscheidung zwischen Arbeit- und Werktage berücksichtigen könnte. Da sind die Lastenprofile ja sehr unterschiedlich.

Alternativ oder zusätzlich: Andocken an eine influxdb - in meinem Fall von evcc. Dort wird das "bereinigte" Lastprofil (Hausverbrauch = Gesamtlast minus Wallbox) bereits mit aufgezeichnet.

Danke Dir und viele Grüße

ohAnd, 2025-03-25:
Hi @WolfImBusch ,

siehe oben "herausgeschnittene" Teilumfänge. ;-)

Influxdb sehe ich derzeit als nicht zielführend, da entweder openHAB oder HomeAssistant jeweils "user friendly" Persistancies mitbringen. (u.a. anderem eben auch InfluxDB im Hintergrund)
Daher würde ich es hier erstmal zurück stellen.

Gruss

WolfImBusch, 2025-11-12:
Hi @ohAnd .

since we have a good and working solution gathering the load profile from Home Assistant/openhub I propose to close this issue here. Maybe we can reopen this point in a discussion in case of someone else come to this point ... 😅

closed 2025-11-13 by ohAnd