ohAnd/EOS_connect #11: Extract "Current Controls" values to Homeassistant entities

PRiekenberg, 2025-04-12:
Hi,

thank you for this nice project!

Is it possible to extract the values of the "Current Controls" Tab to Homeassistant entities via API or MQTT? That would make it possible to control other PV / Batteries via automations in homeassistant based on your controls.

WolfImBusch, 2025-04-12:
This would also help me a lot, especially the distribution via MQTT. Background: I would like to display the current battery mode in the PV dashboard “Solectrus” which mainly operates via MQTT … 
Thank you and best regards!

ohAnd, 2025-04-12:
Should be possible ... but needs some time due to some other topics on my side. I will mark this as a feature request/ enhancement.

Maybe you (@all) can collect here some more details which values should be exposed and maybe also in which clustering.

... additonally I'm a fan of MQTT Autodiscovery (by Homeassistant) so I would combine this...

PRiekenberg, 2025-04-15:
It would be good if we could export all the values under "Current Controls":

Overall State
AC Charge
DC Charge
Discharge Allowed

This would allow us to control our batteries / inverters based on these values.

Also an export of the "Dynamic Max Charge Power" and "Planned Time" for next charge would be nice to have.

ohAnd, 2025-04-15:
Hi @PRiekenberg ,

just fyi ... in the mean time (until mqtt will be integrated) you can already use .../json/current_controls.json to get the current "demands" of EOS for your config.

```
{
    "current_states": {
        "current_ac_charge_demand": 0.0,
        "current_dc_charge_demand": 5000.0,
        "current_discharge_allowed": false,
        "inverter_mode": "MODE_AVOID_DISCHARGE",
        "evcc_charging_state": false
    },
    "battery_soc": 39.1,
    "battery_max_charge_power_dyn": 5000,
    "timestamp": "2025-04-15T18:41:54.943938+02:00",
    "state": {
        "request_state": "response received",
        "last_request_timestamp": "2025-04-15T18:40:34.226026+02:00",
        "last_response_timestamp": "2025-04-15T18:41:54.384392+02:00",
        "next_run": "2025-04-15T18:44:54+02:00"
    }
}
```

Currently used for the dashbaord, but getting instantly the latest values of EOS/ Battery and EVCC

PRiekenberg, 2025-04-18:
Where can I find the json/current_controls.json file? I am using EOS Connect as an Add-on to home assistant. Do you know where the add on files are located?

ohAnd, 2025-04-18:
Hi @PRiekenberg ,

sorry you're right, it is not documented yet...

After starting of EOS-connect, a webserver will be serving the dashboard. This dashboard exchanges data via a small api.
The dashboard is reachable at http://<your_eos_instance_ip>:8081 or if you running EOS-connect via HA addon http://homeassistant:8081.

API description:
- GET /json/optimize_request.json
  - Description: Retrieves the last optimization request.
  - Response: JSON object containing the last optimization request.
- GET /json/optimize_response.json
  -  Description: Retrieves the last optimization response.
  - Response: JSON object containing the last optimization response.
- GET /json/current_controls.json
  - Description: Provides the current system states, including:
    - AC/DC charge demands
    - Discharge allowance
    - Battery state of charge (SOC)
    - Inverter mode
    - EVCC charging state and mode
    - Dynamic max charge power
    - Current optimization state
    - EOS-connect version and timestamp
  - Response: JSON object with detailed system states.

Example for controls: http://homeassistant:8081/json/current_controls.json

PS: controls content is grown over the last days ;-)

closed 2025-04-25 by ohAnd

PRiekenberg, 2025-04-25:
Hi, your commit from today doesn't seem to work correctly.

I get the following error when I try to start your addon after the update to 0.1.10:

Traceback (most recent call last):
  File "/app/src/eos_connect.py", line 25, in <module>
    from interfaces.mqtt_interface import MqttInterface
  File "/app/src/interfaces/mqtt_interface.py", line 16, in <module>
    import paho.mqtt.client as mqtt
ModuleNotFoundError: No module named 'paho'

ohAnd, 2025-04-25:
is already fixed ... forget the dependency for the dockerfile ... 0.1.11

ohAnd, 2025-04-25:
and additionally - that is the first step of implementation ... means the published data to mqtt are ok and maybe not enough ;-), but the controls from outside (override) are only preparations for upcoming features

for further needs/ discussion please create a new issue

commit 1a97a051 (2025-04-25): feature: Add MQTT configuration support and interface for Home Assistant integration - close #11 Extract "Current Controls" values to Homeassistant entities
PR #25 (2025-04-25): Develop
- feature: Enhance EVCC interface and UI:
  - Added detailed vehicle data fetching to the EVCC interface.
  - Updated HTML to display vehicle charging details and status.
  - Improved CSS for smoother transitions on value changes.
- feature: Add MQTT configuration support and interface for Home Assistant integration - close #11 
- fix: (inverter) add version handling for Fronius GEN24 API endpoints