ohAnd/EOS_connect #23: Fronius Firmware 1.36.5-1

danielaussieb, 2025-04-23:
Seit dem neuesten Firmwareupdate gibts folgenden fehler im log beim start

successfully started
2025-04-23 15:39:42 INFO [Main] Starting eos_connect - version: 0.1.9
2025-04-23 17:39:42 INFO [Main] set user defined time zone to Europe/Berlin and loglevel to DEBUG
2025-04-23 17:39:42 INFO [EOS] Getting EOS version: <2025-04-09
Traceback (most recent call last):
  File "/app/src/eos_connect.py", line 101, in <module>
    inverter_interface = FroniusWR(inverter_config)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/interfaces/inverter_fronius.py", line 78, in __init__
    self.previous_battery_config = self.get_battery_config()
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/interfaces/inverter_fronius.py", line 166, in get_battery_config
    response = self.send_request(path, auth=True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/interfaces/inverter_fronius.py", line 506, in send_request
    raise RuntimeError(
RuntimeError: [Inverter] Request 0 failed with 404-Not Found. 
	 path:/config/batteries, 
	params:None 
	headers {} 
	nonce 0 
	payload 
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2025-04-23 15:40:33 INFO [Main] Starting eos_connect - version: 0.1.9
2025-04-23 17:40:33 INFO [Main] set user defined time zone to Europe/Berlin and loglevel to DEBUG
2025-04-23 17:40:33 INFO [EOS] Getting EOS version: <2025-04-09
Traceback (most recent call last):
  File "/app/src/eos_connect.py", line 101, in <module>
    inverter_interface = FroniusWR(inverter_config)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/interfaces/inverter_fronius.py", line 78, in __init__
    self.previous_battery_config = self.get_battery_config()
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/interfaces/inverter_fronius.py", line 166, in get_battery_config
    response = self.send_request(path, auth=True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/src/interfaces/inverter_fronius.py", line 506, in send_request
    raise RuntimeError(
RuntimeError: [Inverter] Request 0 failed with 404-Not Found. 
	 path:/config/batteries, 
	params:None 
	headers {} 
	nonce 0 
	payload 
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

ohAnd, 2025-04-23:
Hi @danielaussieb ,

hab mal im hier gelinkten Branch einen "Quick Hack" anhand der Aussage in https://github.com/muexxl/batcontrol/issues/155 eingesetzt ohne Möglichkeit zu testen und demnach ebenso "vorzupreschen" ;-)

Gerne mal probieren, vielleicht war es 'nur' das ...

-------

Zum An/Abschalten unter Inverter in der config version mit dem u.g. String belegen.

```yaml
# Inverter configuration
inverter:
  type: fronius_gen24  # Type of inverter - fronius_gen24, default (default will disable inverter control - only displaying the target state) - preset: default
  address: 192.168.1.1 # Address of the inverter
  user: customer # Username for the inverter
  password: *********** # Password for the inverter
  max_grid_charge_rate: 5000 # Max grid charge rate in W - default: 5000
  max_pv_charge_rate: 5000 # Max PV charge rate in W - default: 5000
  max_bat_discharge_rate: 5000 # Max battery discharge rate in W (currently not used) - default: 5000
  version: ">=1.36.5-1" # Version of the inverter - default: "old" e.g. 1.35.8-1 and earlier
# EVCC configuration
```

WolfImBusch, 2025-04-25:
Hi @danielaussieb ,

konntest Du den "Quick-Hack" bereits testen? Ich würde gerne "nachpreschen" ... 😅

Viele Grüße!!

danielaussieb, 2025-04-25:
ja, aber hilft nicht - gleicher Fehler

ohAnd, 2025-04-25:
Hab es auf Basis von https://www.photovoltaikforum.com/thread/216690-batcontrol-gen24-batterie-tibber-geld-sparen/?pageNo=36 nochmal angepasst mit automatischer Versionserkennung... (inkl. Image unter https://github.com/ohAnd/EOS_connect/pkgs/container/eos_connect) - die Config Geschichte entfällt damit wieder 

@danielaussieb gerne auch das Log zum Fehler hier (mglw. nochmal) posten auf Basis des neuen speziellen Image "[feature_dev_23-fronius-firmware-1365-1-start-problems](https://github.com/users/ohAnd/packages/container/eos_connect/402663202?tag=feature_dev_23-fronius-firmware-1365-1-start-problems)"

danielaussieb, 2025-04-25:
da brauch ich jetzt hilfe, da ich keinen Plan hab wie ich die version zum laufen bekomme.... :/

ohAnd, 2025-04-25:
@danielaussieb ok, das Update zum Fix  ist von der Art und Weise her, das gleiche wie oben, nur nun zusätzlich mit Image um es eigentlich einfacher zu machen. (kannst aber genauso wie vorher (?) eben auch direkt aus dem branch lokal updaten) 

Sorry, war wohl missverständlich.

Der Fix lag und liegt derzeit erstmal nur hier https://github.com/ohAnd/EOS_connect/tree/23-fronius-firmware-1365-1-start-problems - daher gibt es hier auch noch kein HA addon oder entsprechendes Release - also entweder direkt lokal starten oder eben nun auch per 'docker pull'

WolfImBusch, 2025-04-25:
Hi @ohAnd 

hab nicht nur die neue Fronius SW, sondern auch den Fix installiert - leider scheint der nicht zu greifen ....

Viele Grüße!

```
eos_connect-eos_connect-1  | 2025-04-25 14:18:01 INFO [Main] Starting eos_connect - version: 0.1.38-develop
eos_connect-eos_connect-1  | 2025-04-25 16:18:01 INFO [Main] set user defined time zone to Europe/Berlin and loglevel to DEBUG
eos_connect-eos_connect-1  | 2025-04-25 16:18:01 INFO [EOS] Getting EOS version: >=2025-04-09
eos_connect-eos_connect-1  | 2025-04-25 16:18:01 INFO [Inverter] Current sw revision: {'major': 1, 'minor': 36, 'patch': 5, 'build': 1}
eos_connect-eos_connect-1  | 2025-04-25 16:18:01 INFO [Inverter] Using API prefix: '/api'
eos_connect-eos_connect-1  | 2025-04-25 16:18:01 ERROR [Inverter] Login -0- failed, Response: <Response [404]>
eos_connect-eos_connect-1  | 2025-04-25 16:18:01 ERROR [Inverter] Response-raw: <urllib3.response.HTTPResponse object at 0x7f9d83975c90>
eos_connect-eos_connect-1  | 2025-04-25 16:18:01 ERROR [Inverter] Login -1- failed, Response: <Response [404]>
eos_connect-eos_connect-1  | 2025-04-25 16:18:01 ERROR [Inverter] Response-raw: <urllib3.response.HTTPResponse object at 0x7f9d79b563b0>
eos_connect-eos_connect-1  | 2025-04-25 16:18:01 ERROR [Inverter] Login -2- failed, Response: <Response [404]>
eos_connect-eos_connect-1  | 2025-04-25 16:18:01 ERROR [Inverter] Response-raw: <urllib3.response.HTTPResponse object at 0x7f9d79b56170>
eos_connect-eos_connect-1  | 2025-04-25 16:18:01 INFO [Inverter] Login failed 3 times .. aborting
eos_connect-eos_connect-1  | Traceback (most recent call last):
eos_connect-eos_connect-1  |   File "/app/eos_connect.py", line 101, in <module>
eos_connect-eos_connect-1  |     inverter_interface = FroniusWR(inverter_config)
eos_connect-eos_connect-1  |   File "/app/interfaces/inverter_fronius.py", line 88, in __init__
eos_connect-eos_connect-1  |     self.previous_battery_config = self.get_battery_config()
eos_connect-eos_connect-1  |                                    ~~~~~~~~~~~~~~~~~~~~~~~^^
eos_connect-eos_connect-1  |   File "/app/interfaces/inverter_fronius.py", line 176, in get_battery_config
eos_connect-eos_connect-1  |     response = self.send_request(path, auth=True)
eos_connect-eos_connect-1  |   File "/app/interfaces/inverter_fronius.py", line 514, in send_request
eos_connect-eos_connect-1  |     self.login()
eos_connect-eos_connect-1  |     ~~~~~~~~~~^^
eos_connect-eos_connect-1  |   File "/app/interfaces/inverter_fronius.py", line 600, in login
eos_connect-eos_connect-1  |     raise RuntimeError(
eos_connect-eos_connect-1  |         '[Inverter] Login failed repeatedly .. wrong credentials?'
eos_connect-eos_connect-1  |         )
eos_connect-eos_connect-1  | RuntimeError: [Inverter] Login failed repeatedly .. wrong credentials?
```

WolfImBusch, 2025-04-25:
Hi @danielaussieb ,

@ohAnd hat noch mal eine neue Version der Spezialversion erzeugt - die funktioniert bei mir nun ...

Viele Grüße!!

commit 7353e430 (2025-04-25): Merge pull request #24 from ohAnd/23-fronius-firmware-1365-1-start-problems

fix #23 fronius firmware 1365 1 start problems
closed 2025-04-25 by ohAnd