ohAnd/EOS_connect #27: HA Load and PV Prduction History

stefanriegel, 2025-04-26:
Hi,

I’m working on integrating my Huawei solar system with Home Assistant using your EOS_connect extension. I have a few questions and observations:
	
1.	Load Values: I’m trying to determine the most accurate way to represent my home’s energy consumption. Should I consider the total consumption, including both variable loads and household usage? Or focus solely on the net difference between PV production, variable loads, and grid consumption?

 
2.	PV Production History: Is it possible to retrieve historical PV production data over the past weeks? This would help in analyzing performance trends and optimizing energy usage.

3.	Battery Prediction Anomalies: I’ve noticed that the system predicts my battery will be empty by 1 PM and suggests AC charging in the evening, which doesn’t align with actual usage patterns. Could this be due to how the load values are configured or perhaps an issue with the prediction algorithm?
For context, I’m using a Huawei system integrated into Home Assistant via the Modbus TCP method. History is set to 90 days

Any insights or recommendations you can provide would be greatly appreciated.

Thank you for your support!

stefanriegel, 2025-04-26:
[image]

ohAnd, 2025-04-27:
Hi @stefanriegel ,

to your 1. question:

The "needed" load value is the overall power (wattage) consumption of your household - means every consumer should be here included - independently from which source the energy for the "same" houshold is provided. This have to be stored in a persistance, as Openahb or Homeassistant easily provide per item/ entity.

That means this load value should "automatically" logging your "variable" *and* household usage. ("variable" - I would define it as "regular variable" loads e.g. washingmachine/ dryer maybe often on weekend or something like this)

**EOS connect** take the persisted data of this value and summerize this to an hourly energy consumption according to your consumption. How that is done - see here https://github.com/ohAnd/EOS_connect?tab=readme-ov-file#getting-historical-values

From the current point of view: that is the easiest way to have a more dynamic approach to take care about changes in usage over the year. (e.g. spring/ summer less heating in the evening but more AC over the day or something similiar)

> Or focus solely on the net difference between PV production, variable loads, and grid consumption?

As described above - the net difference between all energy sources and all energy consumers should be "the load".

*Note: especially the big consumer "wallbox" can be substracted by config of another item/ entity with only this load - to have this as extraordinary load and not include in the load prediction to EOS*

2.
Yes it's usually possible - maybe you can open for that a new and more detailling issue to discuss the idea behind. I'm also thinking about something in this direction - more with the idea to get a better setting for the pv prediction (goal: automatically optimized entries for 'horizon')

3. 
Based on your screenshot: In my opinion there is something really wrong with your "Load" - means the hourly values should be never negative. (negative load is implicitly an energy source) I would assume this connects with your 1. question - that your currently not providing the right data ;-) - means when you fix that - then the chart should be more understandable at first and also calculated recharges, discharge command, soc and so on makes more sense.
To clarify what **EOS connect** sends to EOS and what the repsonse is - you can use the Dashabord API
-> http://homeassistant:8081/json/optimize_request.json
-> http://homeassistant:8081/json/optimize_response.json
Check in the request, what is given here for 'gesamtlast':
```json
{
  "ems": {
    "pv_prognose_wh": [0, ..., 0.0003205],
    "einspeiseverguetung_euro_pro_wh": [0, ..., 0],
    "preis_euro_pro_wh_akku": 0,
    "gesamtlast": [396.8975, 301.25305, 300.03685, 257.6488, 268.6918, 290.91955, 299.53855, 326.11335, 353.7833, 383.1619, 514.63665, 1007.0328, 1534.64875, 724.4491, 1648.8038, 1108.05115, 772.87125, 665.93295, 1110.42845, 704.88145, 526.4264, 446.4665, 498.67215, 420.9993, 328.194, 290.7371, 301.52515, 297.66605, 274.8474, 256.51125, 386.3983, 358.5091, 375.92015, 350.5336, 631.8915, 650.22295, 946.81305, 922.6812, 990.7707, 785.56225, 500.1753, 487.29445, 446.2675, 434.1165, 527.7583, 432.62245, 796.35395, 469.46665]
  },
...
```
This is the result of the gathering of the given data... (above as an example)

stefanriegel, 2025-04-30:
i change it from Grid meter (it was my flout, you're right) to my EVCC Household Load Sensor, but the negative value are still there. Maybe we can move this to a discussion Thread

stefanriegel, 2025-04-30:
okay i checked the json files:

```
"gesamtlast": [
            -4083.7472,
            -4193.66395,
            -4191.7027,
            -4231.57285,
            -4194.41855,
            -4201.1754,
            -4225.1212,
            -3823.9464,
            -3603.3584,
            -1977.52135,
            -3618.25335,
            -3933.94145,
            -2116.8833,
            -3293.06235,
            -2804.4591,
            -3749.7058500000003,
            -1375.35315,
            -1883.5673499999998,
            -1690.4801000000002,
            -3894.0186999999996,
            -4069.3296,
            -3970.4190999999996,
            -3521.0941,
            -4098.3674,
            -4297.745999999999,
            -4318.081249999999,
            -4201.60585,
            -4237.9479,
            -4246.132949999999,
            -4283.8659,
            -4222.1789,
            -4131.0085,
            -3260.3827,
            -2871.18565,
            -2405.5718500000003,
            -5769.501850000001,
            -6678.5302,
            -6577.00325,
            -6670.4377,
            -6847.51485,
            -6791.2365,
            -5601.37015,
            -4190.55055,
            -4094.0754000000006,
            -3560.9197,
            -3728.41765,
            -4047.34985,
            -1137.3663

```

and my sensor is (its never negative):

sensor.evcc_home_power

with:

[image]

danielaussieb, 2025-04-30:
here are a compareable problem, im also using the evcc home power sensor
the request have some negative values - but not from the sensort itself

  "gesamtlast": [
            471.1214,
            480.3884,
            462.6747,
            472.4696,
            466.7747,
            481.921,
            286.7387,
            424.4824,
            163.4671,
            2812.2867,
            4797.3298,
            2495.2172,
            1094.6676,
            655.1047,
            1125.8066,
            1491.0849,
            1732.7559,
            -4713.3167,
            -7506.0092,
            -7060.1259,
            746.2653,
            604.3645,
            497.1838,
            479.2616,
            474.9593,
            470.2085,
            472.7569,
            483.2318,
            506.0662,
            514.0913,
            365.239,
            384.7908,
            162.8909,
            385.1312,
            2223.0581,
            4008.9608,
            3450.8078,
            1383.4491,
            465.3537,
            936.3058,
            720.2852,
            3565.2608,
            1790.4897,
            688.1055,
            588.3123,
            567.3158,
            482.1863,
            461.9881

[image]

ohAnd, 2025-05-01:
Hi @stefanriegel and @danielaussieb ,

please check also the logfile with loglevel = debug... here you will see which detail data will be used for the "gesamtlast" in the request. 
Looks like this:
```
2025-05-01 10:11:57 DEBUG [LOAD-IF] Energy for 2025-04-17 03:00:00+02:00: 300.8 Wh (car load:   0.0 Wh)
2025-05-01 10:11:57 DEBUG [LOAD-IF] Energy for 2025-04-17 04:00:00+02:00: 257.9 Wh (car load:   0.0 Wh)
2025-05-01 10:11:57 DEBUG [LOAD-IF] Energy for 2025-04-17 05:00:00+02:00: 298.5 Wh (car load:   0.0 Wh)
```

Do you already configured the ecar charging laod? e.g.
```yaml
load_sensor: sensor.household_power
=> car_charge_load_sensor: sensor.evcc_mqtt_evcc_charge_power
```

Data from this will be substratcted from household load to calculate the value for each hour. Due to some sporadic failures that on my side evcc didn't serve the right data for persisting within in HA I also extended a primitive error recognition see [commit](https://github.com/ohAnd/EOS_connect/commit/9c5d03f841f4813e45667cbd2f8bb7f8235f2082#diff-fd65c2d186bde49517c3267f4f33d0288418d49a2f9530a6449b9f8ae461611f) to prevent sporadic negative values if car load has to high and therfore wrong data 
```python
if car_load_energy < energy:
                energy = energy - car_load_energy
else:
 # take the household energy only
 ....
```

Please check with this info your source data... At the end only positive values have to be appear in "gesamtlast"...

stefanriegel, 2025-05-01:
your last update fixed my issue, idk why

MatzeBu, 2025-05-04:
`2025-05-04 20:36:00 ERROR [LOAD-IF] DATA ERROR load smaller than car load - Energy for 2025-04-28 13:00:00+02:00: 230.1 Wh (car load: 2097.0 Wh)
2025-05-04 20:36:00 DEBUG [LOAD-IF] Energy for 2025-04-28 13:00:00+02:00: 230.1 Wh (car load: 2097.0 Wh)
2025-05-04 20:36:00 ERROR [LOAD-IF] DATA ERROR load smaller than car load - Energy for 2025-04-28 14:00:00+02:00: 236.3 Wh (car load: 2092.2 Wh)`

I have this problem: The Car load is not 2097.0Wh. it is 2,097Wh. The standby power of my wallbox.
I created a new sensor in home Assistent in kwh. But it needs some days, because it isn't old enogh...

ohAnd, 2025-05-05:
Got it...
My "bad" - is an automatic detection from the early days ... if the car laod comes in kW ... 
With standby power goes this kind of implementation in the wrong direction ... 

So, cleaned this ... please check the latest develop.

=> detection is removed and the car load will be always expected in watts

closed 2025-05-05 by ohAnd