ohAnd/EOS_connect #34: Add other Loads (Heat pumps, Dishwasher, Washing machine)

PRiekenberg, 2025-05-09:
Hi, thank you for your nice work!

Is it possible to add other loads for the prediction like Heat pumps, dishwashers or washing machines?

It would be nice to have the next start date and time for the next dishwasher run or heat pump heat cycle. These values could be written back to home assistant via MQTT and then used in automations.

I think the EOS already supports these devices types.

ohAnd, 2025-05-25:
I've played a little bit with that thing ...
And integrated for the first tries, the following:
- config has new part at load:
```yaml
additional_load_1_sensor: "additional_load_1_sensor", # item / entity for wallbox power data in watts. Leave empty if not used.
additional_load_1_runtime: 2 # runtime for additional load 1 in minutes - default: 0 (or empty) = not used
additional_load_1_consumption: 1500 # consumption for additional load 1 in Wh - default: 0 (or empty) = not used
```
- first param is useful to substract also this not regular load from the historic data to have a more accurate predicition, but not needed to give the following 2 params to EOS
- runtime and consumption is the need by EOS optimize request - means how long the homepliance should be running and which energy will bs consumed in that time (overall not per hour)

this will lead to 2 new mqtt outputs:
- EOS Home Appliance Released -> should the home appliance (dishwasher) start now yes/no given by optimisation result
- EOS Home Appliance Start Hour -> the optimized start hour in the next 48h - current hour of the day (until tomorrow end of day) given by optimisation result

_side note:
Currently, based on EOS web API, there is only one additional consumer possible. EOS itself seems has internally more possinbilites incl. specific starthour from when in the day the optimization should be taken the add load into account. So, here are currently some limits._

open:

Currently the optimization will take this info with every request - means it will be assumed that this "wish" is given for the next hours over whole day starting with every request again
-> so it has to be discussed
- in which "time frame" this request should be included - means specific start hour of the day and so the request will given here - or something like that ... 
- Additionally to that, let's assume that is a dishwasher with 3 h and 1500 Wh for that time - EOS will give back e.g. start at 3pm running until 6 and consuming 500Wh from 3-4, 500Wh from 4-5, ... -> Now it is 4 o'clock - EOS connect/ EOS should be know if the dishwasher was already started at 3 and giving only the remaining time and energy to EOS - to get the right feedback ...

Thus, the current implementation is aiming 'playing around with this' and not the final target. ;-)

MatzeBu, 2025-09-24:
I'm also intrested in such a solution... it's tricky and not so easy...
I have a Heat Pump for hot water. If there is much cheap energy i like to heat up to 70°C. If it's not so much energy there then in the cheapest hours to normal 50°C.

My first try in EOS configuration I set:
```
additional_load_1_sensor: sensor.brauchwasser_warmepumpe_switch_0_power
additional_load_1_runtime: 3
additional_load_1_consumption: 1600
```

This is the result:
[image]

[image]

I don't understand the result. There is no sense to start to these times where the price is high and the sun low..

I would change the runtime and consumption to a sensor from home assistant which caluclated with the mean over 14 days. But the big problem for first is, that it don't calculate good times...

ohAnd, 2025-09-26:
hmm... ;-)

Maybe you have a misundertstanding here, how the optimzation works.

Main idea for EOS as of my understanding:

You have different sources of energy with different prices, grid with your provider prices, battery with already charged energy (can be also set with cost) and "no-cost" PV energy and on the other side consumers ... 
For the needed energy of the consumers eos try to find the best fitting source at a certain time.

Your idea with the heatpump, to warm up the water, based only on the price will be not directly given here.
From EOS perspective the heatpump is an addtional consumer. With your config you give EOS the task optimize this additional load in addtion to your hosuehold loads, but based on the same energy sources.

So your charts showing directly that - at the projected times there is enough "cheap" energy already in the battery - so the battery source will be used to drive your houshold AND the additonal load at the same time - and the grid energy prices are not relevant at this time.

I guess you want more something similiar to a "surplus control" for your heatpump with different levels based on "cheap energy" with 50/70 target temperatures...

---

What is the idea to store the cheap energy into the water storage (one way), instead of storing the cheap energy into the battery and using this at the time you have to heat up the water?

MatzeBu, 2025-09-27:
Thank you very much!
OK, I think and see that EOS uses the best sources for the planned time. 
But which way is the projected time calculated? If the heat pump work before the batterie is full it should be cheaper than work with energy from the batterie. 
And because its cheaper too use the energy in this moment, there is something like a surplus control. (In the first step without different temperature levels)
The reason for heating the water in the cheapest time is, that I can use the energy from the batterie in another way. And if the heatpump work for 1-3h, its enough for the day. 

Another thing: My thinking was that I give the power sensor in W from the additional load and EOS calculate the difference between additional load consumption and the needed energy from this day... 
The question in another way: in which way use EOS the additional_load_1_sensor?

Another idea:
If it is possible to add additional loads to a specific time or time area and calculate the best time. Is there a chance to use the loading plan from evcc? Give the plan from evcc to EOS and EOS calculate with this additional load?

ohAnd, 2025-09-27:
;-) Yep, that's the problem with the current EOS version and additional load also for me. I can't see something, how the time frame for the additonal load can be set. EOS uses the duration and amount put that in the optimization time frame of 48 hours. But I see currently nothing how we can say... start with earliest in 6 hours or at 7 am ... 
Needs to be discussed at EOS side / issue or discussion.

evcc charging plan is seperated and in this way included to not "disturb" the optimization ... possible additional loads from evcc will be leading to the same problem as discussed above ...

but could be also more eximined, how a "contributional" optimization together with evcc is useful/ possible

> Another thing: My thinking was that I give the power sensor in W from the additional load and EOS calculate the difference between additional load consumption and the needed energy from this day...
The question in another way: in which way use EOS the additional_load_1_sensor?

historic data from additonal load will be substracted from overall consumption for the prediction - with the idea that this energy consumer will be used variable and is not part of repeated main load

ohAnd, 2026-09-28:
closed with #55

closed 2026-09-28 by ohAnd