ohAnd/EOS_connect #39: Error Load smaller then car load, if both values are 0.0 Wh

PRiekenberg, 2025-05-13:
Hi, I get the following error:

2025-05-13 11:21:29 ERROR [LOAD-IF] DATA ERROR load smaller than car load - Energy for 2025-05-07 05:00:00+02:00:   0.0 Wh (car load:   0.0 Wh)

Could you please adjust load_interface.py to catch the case, when both values are 0.0 Wh? 

Thank you!

Kind regards

ohAnd, 2025-05-13:
Oha ... ;-) weird use case ... I would assume at least the "machine" which is logging the household load should be part of this and consume something...

But, ok ... It's fixed.

ohAnd, 2025-05-14:
Hi @PRiekenberg,

same prob @WolfImBusch side - after some investigation on his side there was a leakage of load data due to an "longer" running update homeassistant instance (problems with updateing 2025.05.1).

Means there was a 3 hour 0 wattage for load and this was an addtional bug that this data were skipped for the request array.

It's currently fixed on develop - and will be soon on main...

PRiekenberg, 2025-05-14:
Thank you for this fix!

For me it was the tibber integration which is used to calculate my load, which was not available for a couple of hours, resulting in the load of 0.

Nice to hear this is fixed, so that eos connect can handle that.

commit 2da0d729 (2025-05-13): fix: adjust energy comparison logic to allow equal values for car and load energy - fixes #39 Error Load smaller then car load, if both values are 0.0 Wh
closed 2025-05-17 by ohAnd