ohAnd/EOS_connect #46: The load sensor readout seems to be broken

paule96, 2025-05-21:
I currently try to use EOS connect in my home assistant installation.

I have configured the load settings as following:

```yaml
source: homeassistant
url: http://homeassistant.local:8123/
load_sensor: sensor.home_consumption_eos_connect
car_charge_load_sensor: sensor.home_consumption_eos_connect
access_token: eyJ.....
```

> Ignore the car charge load sensor because I just used it, to see if this fixes anything.

but when I wait for the optimization calculation I get the following result for the load:

[image]

So a predication of less then 10 watts.

The sensor in homeassistant looks like that:

[image]

So way higher than the prediction. And that always. (I actually tried here already with higher values because the sensor is currently my real consumption times 10)

I think something is off here.

> Note: What I also noted is, that optional parameters like the car_charge_load can not be removed.

paule96, 2025-05-21:
Also I discovered a weird logging:

[image]

Looks like he tries to predict the past. Like a week in the past.

ohAnd, 2025-05-21:
Please check the readme -> https://github.com/ohAnd/EOS_connect?tab=readme-ov-file#collecting-data
I think that explain why you see something "wierd" logging ...

And especially also -> https://github.com/ohAnd/EOS_connect?tab=readme-ov-file#home-assistant-persistance regardiung HA persistance

paule96, 2025-05-21:
Ah I see. So if you have less than 10 days of data this is basically not working?
This was not so clear from the documentation side.

But now I see data if I use another sensor that exists a bit longer. Thank you very much. Should I open a new issue to make some settings more optional?

ohAnd, 2025-05-21:
Yes, then we can close here.

closed 2025-05-22 by ohAnd