ohAnd/EOS_connect #53: Feed In Prices are not shown in Income

thecem, 2025-05-31:
EOS: https://github.com/Duetting/ha_eos_addon/tree/main/eos
EOS_Connect: latest develop 0.1.0.76-develop (0.1.0.77-develop not available for update right now)

I would expect the income shown by a multiply of feed_in_price, but the income is 0.
Ist there a configuration fault?

[image]

<details>

<summary>HA Addon Config</summary>

``` yaml
load:
  source: homeassistant
  url: http://192.168.1.100:8123
  access_token: >-
    1234
  load_sensor: sensor.home_power
  car_charge_load_sensor: sensor.car_charge_load_sensor
  additional_load_1_sensor: sensor.additional_load_1
  additional_load_1_runtime: 3
  additional_load_1_consumption: 1500
eos:
  server: 192.168.1.100
  port: 8503
  timeout: 360
price:
  source: fixed_24h
  token: tibberBearerToken
  fixed_24h_array: >-
    15.07, 15.07, 15.07, 15.07, 15.07, 28.17, 28.17, 28.17, 28.17, 28.17, 28.17,
    28.17, 28.17, 28.17, 28.17, 28.17, 28.17, 28.17, 28.17, 28.17, 28.17, 28.17,
    28.17, 28.17
  feed_in_price: 0.0755833
  negative_price_switch: false
battery:
  source: homeassistant
  url: http://192.168.1.100:8123
  soc_sensor: sensor.state_of_charge_fronius_storage_0_http_fg2410ak3l_fritz_box
  access_token: >-
    1234
  capacity_wh: 28480
  charge_efficiency: 0.88
  discharge_efficiency: 0.88
  max_charge_power_w: 10000
  min_soc_percentage: 10
  max_soc_percentage: 100
  price_euro_per_wh_accu: 0.0001
pv_forecast:
  - name: balkon_dg
    lat: 50.0
    lon: 8.0
    azimuth: 52
    tilt: 60
    power: 900
    powerInverter: 1000
    inverterEfficiency: 0.9
    horizont: ""
  - name: balkon_og
    lat: 50.0
    lon: 8.0
    azimuth: 52
    tilt: 60
    power: 900
    powerInverter: 1000
    inverterEfficiency: 0.9
    horizont: ""
  - name: dach_nord_ost
    lat: 50.0
    lon: 8.0
    azimuth: -128
    tilt: 37
    power: 4620
    powerInverter: 4200
    inverterEfficiency: 0.9
    horizont: ""
  - name: dach_nord_west
    lat: 50.0
    lon: 8.0
    azimuth: 142
    tilt: 37
    power: 5390
    powerInverter: 4200
    inverterEfficiency: 0.9
    horizont: ""
  - name: dach_sud_ost
    lat: 50.0
    lon: 8.0
    azimuth: -38
    tilt: 37
    power: 5390
    powerInverter: 10000
    inverterEfficiency: 0.9
    horizont: ""
  - name: dach_sud_west
    lat: 50.0
    lon: 8.0
    azimuth: 52
    tilt: 36
    power: 1540
    powerInverter: 1500
    inverterEfficiency: 0.9
    horizont: ""
  - name: garage_dach
    lat: 50.0
    lon: 8.0
    azimuth: 52
    tilt: 35
    power: 1400
    powerInverter: 1200
    inverterEfficiency: 0.9
    horizont: ""
  - name: garten_wand
    lat: 50.0
    lon: 8.0
    azimuth: 52
    tilt: 21
    power: 350
    powerInverter: 300
    inverterEfficiency: 0.9
    horizont: ""
  - name: garten_haus
    lat: 50.0
    lon: 8.0
    azimuth: -24
    tilt: 21
    power: 1200
    powerInverter: 1200
    inverterEfficiency: 0.9
    horizont: ""
  - name: lounge_zaun
    lat: 50.0
    lon: 8.0
    azimuth: -24
    tilt: 90
    power: 1050
    powerInverter: 1050
    inverterEfficiency: 0.9
    horizont: ""
  - name: terrasse
    lat: 50.0
    lon: 8.0
    azimuth: 52
    tilt: 9
    power: 2800
    powerInverter: 2800
    inverterEfficiency: 0.9
    horizont: ""
  - name: wand
    lat: 50.0
    lon: 8.0
    azimuth: -38
    tilt: 90
    power: 7500
    powerInverter: 7500
    inverterEfficiency: 0.9
    horizont: ""
inverter:
  type: fronius_gen24
  address: 192.168.1.132
  user: customer
  password: Asfdgh.12
  max_grid_charge_rate: 9000
  max_pv_charge_rate: 10000
  max_bat_discharge_rate: 9000
evcc:
  url: http://192.168.1.100:7070
mqtt:
  enabled: true
  broker: localhost
  port: 1883
  user: mqtt
  password: xxxx
  tls: false
  ha_mqtt_auto_discovery: true
  ha_mqtt_auto_discovery_prefix: homeassistant
refresh_time: 6
time_zone: Europe/Berlin
eos_connect_web_port: 8081
log_level: debug

```

</details>

ohAnd, 2025-06-01:
yes, it should be directly displayed if given by eos response.

---

but ... I see with the last changes around pricing there is currently a bug with wrong naming of config param...

it should be

```yaml
feed_in_price: 0.0755833
```

but currently it will be 
```yaml
feed_in_tariff_price: 0.0755833
```

expected

I will change back to `feed_in_price`

btw: nice "not so small" setup ;-)
btw2: do you have an idea what is at 1/2 pm - that there is a battery charge planned/ shown

thecem, 2025-06-01:
btw: thx!

btw2: this red one is the appliance “additional load 1”

[image]

Is there some same for the expense?

ohAnd, 2025-06-01:
You're right... I have to change the color for add load...

Expense comes directly from the optimization... There is no error given, in this context.

thecem, 2025-06-01:
Working. thx!

Is there a plan to add a second loadpoint? Only the first one is shown.

ohAnd, 2025-06-06:
before closing - I saw maybe a missunderstanding on my side ...

> Is there a plan to add a second loadpoint? Only the first one is shown.

Do you mean with 'loadpoint' the evcc incoming data for ecar charging ? ... and then if there 2 or more active at the same time?

Sorry, was too fast and I meant then the addotional loads ...

closed 2025-06-06 by ohAnd

thecem, 2025-06-06:
Yes, a second load point from evcc.