ohAnd/EOS_connect #55: Different load profiles based on binary sensor state

PRiekenberg, 2025-06-04:
Hi,

I am using a heat pump for heating and also for cooling. If the cooling is active, the heat pump needs a lot more energy. 

If my load profile is based on the last 14 days and I turn on the cooling, the predictions is wrong, because my load is much higher.

Is it possible to choose different load profiles based on the state of a specific binary sensor?

ohAnd, 2025-06-04:
Hi,

first thought --- too specific - imagine some other want to connect this with outside temperature, next with room temperature or on sunday is washing day ;-) and there is a specific thing and load for this and so on

but in second: dynamic load "injection" - means an API endpoint (http/mqtt) that can take an hourly consumption value as avg for the day or for 24 h as an string array - this can be dynamically influenced from outside

So, with your specific use case you can put to this endpoint either 0, if heatpump is in heating mode or e.g. 1.5 kWh as a avg value for the whole day or 0,0,0,0,1.2,1.4,1.5,1.6,1.5,1.4, ... and so on for 24h ...

under the hood I would easily reduce the given avg household load by this given "constant" values for this day (and next day) to have finally an adpated load prediciton for the optimization

---
With this, the dynamic should be possible in dependency to external specific circumstances by different user needs.

---

Or you have maybe further ideas how we can define it more generally .

j4rvisstant, 2026-08-27:
+1 for the dynamic load injection endpoint proposed above — it would solve a
problem that also comes up in #201 and #34.

A load forecast built from the same weekday 7 and 14 days ago cannot follow a
cold snap, so with a heat pump the forecast is systematically too low exactly
when battery planning matters most. Putting the heat pump into
additional_load_1 makes it worse, since that subtracts it from the base load
and treats it as a schedulable appliance.

The data needed to do better already exists on the Home Assistant side: outdoor
temperature forecast, heating curve, and measured heat pump consumption per
temperature bin. What is missing is only the way to hand the resulting profile
to EOS Connect. An endpoint accepting a 24 h (or 96-slot) array as you sketched
would let everyone plug in their own model, without EOS Connect having to know
anything about heat pumps, cooling, or washing days.

Happy to test a prototype against a real heat pump installation this winter.

ohAnd, 2026-09-28:
hi together,

maybe you noticed already that there is a new config section for so called "managed loads" ...

some detailing doc: https://ohand.github.io/EOS_connect/develop/user-guide/configuration.html?level=expert#managed-loads
(will be more structured soon)

That new feature needed "some" iterations ( see above ;-) ) - there are several profiles and my own need for pool heat pump was tested and corrected over weeks ...
The other profiles are mostly dervied from that iterations.
additionally 2 open profiles:
- external contigent
- external profile

The external profiles are what we discussed above.

Will be nice if you give those profiles a try - except pool heatpump - nothing could be tested with real loads ...

hp-8472, 2026-09-28:
Thanks a lot. The external load profile is exactly what we had in mind. Plan on our side: profile source timeseries, fed by a Home Assistant template sensor that turns the outdoor temperature forecast into heat pump consumption per slot, based on measured consumption per temperature bin. Heating season hasn't started here yet, so the model needs a few weeks of heating data first; we'll test it then and report forecast vs. actual. Is managed loads planned for the next stable add-on release?

ohAnd, 2026-09-28:
yes, planned ;-)

and for the heat pump as source for house heating with small or big buffer ... maybe we can discuss you 'local' solution then further as specific profile inside eos_connect - after some tests and requirements gathering on your side ;-)

tepebox, 2026-09-29:
OK, I guess I partially understand... But in the end, we (or at least I) have a year of heatpump data... So why can't we derive similar weather situations (temperature, sunny or not, ...) from the history and find out what the heatpump consumed? Sorry, sometimes simplifying challenges to much, I know :-)

closed 2026-09-29 by ohAnd

tepebox, 2026-09-30:
Did some research... But didn't try it yet and I'm a bit hesitant because I still think that I have some gaps in my HA and EOS Connect knowledge ;-) I found this: https://github.com/isaacjmannion/ha-power-predictor. Wouldn't it be a good idea to combine that with EOS Connect? So as far as I understand it would fill the gap I see, analyzing historic data to find out what power consumption i.e. my heat pump will have given the data I have around temperature and consumption in combination with weather forecast... And then it should deliver it as Managed Load to be integrated into EOS Connect, right? @ohAnd  What do you think?