ohAnd/EOS_connect #58: Version 0.1.18 not starting

PRiekenberg, 2025-06-07:
Hi, the homeassistant addon is not starting with version 0.1.18, because it is missing some configs.

When i try to edit config and save, I get the following error:

"Fehler beim Speichern der Add-on Konfiguration, Missing required option 'fixed_24h_array' in EOS connect (855a9a00_eos_connect)"

ohAnd, 2025-06-07:
Jep, unfortunately...
I have to find out the logic in HA addon config for 'optional'...

Short workaround :

Remove 'fixed_24h_array' just before save config. And if other params complaining also.

Then it takes over. And can be started.

closed 2025-06-07 by ohAnd

PRiekenberg, 2025-06-08:
Hi, it is still not working, even with the fix. With default null is set for the variable fixed_24h_array.

When I set double quotes I can save the config, but the add on is not starting and throws the following error:

"/app/src/interfaces/price_interface.py", line 92, in __init__
    float(price) for price in self.fixed_24h_array.split(",")
    ^^^^^^^^^^^^
ValueError: could not convert string to float: ''

ohAnd, 2025-06-08:
thx for the info.

is the second layer ... and targeting #52 

I'm on it.

ohAnd, 2025-06-09:
thx
jep, it's right (for checking the reachability) - but here also 
```yaml
evcc:
  url: "" # URL to your evcc installation, if not used set to "" or leave as http://yourEVCCserver:7070
```

or
```yaml
evcc:
  url: http://yourEVCCserver:7070 # URL to your evcc installation, if not used set to "" or leave as http://yourEVCCserver:7070
```

as setting - will skip the config for evcc

I will check to have more clarity with the given log messages, on second level. In addon config there is this way already given.

ohAnd, 2025-10-01:
Hi,

in HA config both entries are working to switch off:
```yaml
price:
  source: tibber
  token: xxx
  fixed_24h_array: ""
  feed_in_price: 0.0794
  negative_price_switch: true
```
(this one will be automatically taken over as "0" HA)

or
```yaml
price:
  source: tibber
  token: xxx
  fixed_24h_array: "0"
  feed_in_price: 0.0794
  negative_price_switch: true
```

Can you please show your price config in yaml?

ohAnd, 2025-10-01:
add:
with the new HA config gui also for second level of config, you will only need to leave the field empty... 

[image]

then it will be written to 
```yaml
fixed_24h_array: ""
```
in background