ohAnd/EOS_connect #66: Haus-Batterie wird nicht gesperrt, sofern evcc mode = "PV" und Ladenplan aktiv

WolfImBusch, 2025-06-23:
Hi @ohAnd ,

folgende Situation
- evcc mode = "PV"
- evcc Ladeplan erstellt für Abfahrt morgen 6.00h -  FZG 40% SOC haben
- aktuell ist der Preis am günstigsten, so dass evcc sofort mit dem Laden beginnt

Was passiert?
- Laden des Fahrzeugs startet mit 11 kW
- EOS_connect erkennt evcc mode = "PV" und sperrt die Haus-Batterie nicht
- das Laden des Fahrzeugs erfolgt über die Haus-Batterie (4.6 kW) - der Rest kommt aus der Haus-Batterie (5.6 kW)

Erwartetes Verhalten **(t.b.d.)**
- EOS_connect sperrt die Haus-Batterie, wenn Ladeplan ausgeführt wird (wenn das von "außen" überhaupt erkannt werden kann)

Anbei ein paar Bilder, die das veranschaulichen ...

Viele Grüße

[image]

[image]

[image]

ohAnd, 2025-06-23:
Hi @WolfImBusch ,

bisherige Annahme Modus PV und günstiger Preis heisst im evcc api state ( http://<your_evcc_host>:7070/api/state ) im
`loadpoints`
-> `smartCostActive`= true 
und
-> `mode` = pv 
- das ist die Grundlage das bei PV und günstiges Schnellladen die Batterie gesperrt wird.

Daher bitte die Situation nochmal nachstellen und den api/state (nur der betroffende loadpoint) hier posten, um die passenden Abhängigkeiten und scheinbar Unterschiede bei geplanten Laden erkennen zu können.

ohAnd, 2025-07-26:
Hi @WolfImBusch ,

konntest Du hier schon das Problem nochmal erzeugen und den api/state ziehen, das würde mir helfen das Problem einzugrenzen bzw. zu verstehen.

WolfImBusch, 2025-10-16:
Hu @ohAnd ,

hab Dir mal meine komplette /api/state per DM geschickt .. Du kannst das leicht nachvollziehen, indem Du Dir selbst einen Ladeplan mit evcc erstellst, der so definiert ist, dass dieser sofort gestartet wird ..

Viele Grüße

ohAnd, 2025-10-19:
> ...Habe hier gerade ein ähnliches Verhalten. Im EOS Connect state sieht man "avoid discharge evcc fast" aber per mqtt steht der Status immer noch auf "discharge allowed". Ggf schaue ich heute Abend selbst noch mal in den Code.

@paulelsner 

Um zu verstehen, ob ähnliches Problem oder ein einfacher Fehler in Bezug auf Ausgabe MQTT bitte in dem Moment die WebApi -> /json/current_controls.json checken und ggfs. hier mal zeigen.

Denn laut Deinem Screenshot bist du im smart cost fast mode d.h. pv oder pvmin aktiv und smartcostactive in evcc (also gerade der von dir angegebene günstige Preis ist aktiv und damit wird schnell geladen) - Und die Anzeige in der UI basiert auf diesem JSON.

Bitte auch die MQTT states
```
eos_connect/control/overall_state
eos_connect/control/eos_ac_charge_demand
eos_connect/control/eos_dc_charge_demand
eos_connect/control/eos_discharge_allowed
```
zeigen, denn hier wird weiterhin der demand seitens EOS angezeigt, da EOS ja keinen evcc zustand kennt...
unter overall_state solltest du in diesem Zustand (Smart Cost Fast) hier eine '3' sehen. Denn hier soll ja EOS_connect aktiv eingreifen und den EOS Demand überschreiben.

paulelsner, 2025-10-26:
[image]

[image]

```
{
    "current_states": {
        "current_ac_charge_demand": 0,
        "current_dc_charge_demand": 5000.0,
        "current_discharge_allowed": true,
        "inverter_mode": "MODE AVOID DISCHARGE EVCC FAST",
        "inverter_mode_num": 3,
        "override_active": false,
        "override_end_time": 0
    },
    "evcc": {
        "charging_state": true,
        "charging_mode": "now",
        "current_sessions": [
            {
                "connected": true,
                "charging": true,
                "mode": "now",
                "chargeDuration": 9600,
                "chargeRemainingDuration": 1648,
                "chargedEnergy": 28480,
                "chargeRemainingEnergy": 4924.375,
                "sessionEnergy": 28480,
                "vehicleSoc": 71.597,
                "vehicleRange": 243,
                "vehicleOdometer": 55143,
                "vehicleName": "ID.3 Pro Performance",
                "smartCostActive": false,
                "planActive": false
            }
        ]
    },
    "battery": {
        "soc": 39.0,
        "usable_capacity": 3515.0,
        "max_charge_power_dyn": 5000,
        "max_grid_charge_rate": 5000
    },
    "inverter": {
        "inverter_special_data": null
    },
    "localization": {
        "currency": "EUR",
        "currency_symbol": "\u20ac",
        "currency_minor_unit": "ct"
    },
    "state": {
        "request_state": "response received",
        "last_request_timestamp": "2025-10-26T19:29:09.103647+01:00",
        "last_response_timestamp": "2025-10-26T19:30:03.280556+01:00",
        "next_run": "2025-10-26T19:33:03.281872+01:00"
    },
    "eos_connect_version": "0.2.25.153-develop",
    "timestamp": "2025-10-26T19:31:18.626405+01:00",
    "api_version": "0.0.2"
}
```

[image]

Hier noch hoffentlich alle Daten zusammen.

Meine Automatisierung um die Batterie umzuschalten hängt an `select.eos_connect_current_state` und dieser ändert sich hier leider nicht.