ohAnd/EOS_connect #75: Battery SOC drains without PV charging accounted and AC charge triggered despite PV surplus

lieblinger, 2025-07-04:
Hi,

First of all, thank you very much for this great project – I really appreciate all the work you have put into EOS Connect!

I have configured EOS Connect with my Home Assistant sensors, and I am experiencing an issue where the battery State of Charge (SOC) continuously drops over time, even when there is sufficient PV production available. Additionally, AC grid charging is being triggered although there is PV power that should be used for charging the battery.

It seems the PV forecast is not properly taken into account in the SOC estimation and forecast calculations. The forecast graph always shows expenses and never any income, despite actual PV generation.

Below is my configuration for reference:

---

**Configuration:**

```
Load Settings:
source: homeassistant
url: http://localhost:8123
access_token: XXX
load_sensor: sensor.deye_load_power
car_charge_load_sensor: sensor.wallbox_total_charging_power
additional_load_1_sensor: ""
additional_load_1_runtime: 0
additional_load_1_consumption: 0

EOS Settings:
server: localhost
port: 8503
timeout: 180

Price Settings:
source: fixed_24h
token: ""
fixed_24h_array: >-
  19.00, 19.00, 19.00, 19.00, 19.00, 32.00, 32.00, 32.00, 32.00, 32.00, 32.00,
  32.00, 32.00, 32.00, 32.00, 32.00, 32.00, 32.00, 32.00, 32.00, 32.00, 32.00,
  32.00, 32.00
feed_in_price: 0.058
negative_price_switch: false

Battery Settings:
source: homeassistant
url: http://localhost:8123
soc_sensor: sensor.deye_battery_state_of_charge
access_token: XXX
capacity_wh: 168000
charge_efficiency: 0.92
discharge_efficiency: 0.92
max_charge_power_w: 46080
min_soc_percentage: 10
max_soc_percentage: 100
price_euro_per_wh_accu: 0

Inverter Settings:
type: evcc
address: ""
user: ""
password: ""
max_grid_charge_rate: 10000 # 46080
max_pv_charge_rate: 10000 # 46080
```

---

**Observed Behavior:**

* The SOC is continuously decreasing over time and only AC charge increasing it.
* No forecasted income is displayed in the chart.
* AC grid charging starts unnecessarily even when PV is producing more than the consumption load.

---

**Expected Behavior:**

* PV production (and forecast) should be correctly subtracted from the load and used to charge the battery.
* Forecast visualization should show expected income when PV production exceeds consumption.
* AC grid charging should only occur if PV and battery cannot cover the load.

---

**Additional Information:**

* All Home Assistant sensors provide values in watts (not kWh).
* PV forecast is configured and available.
* Please see the attached screenshot of the EOS Connect dashboard showing the issue:

[image]

---

**Questions:**

* Is there any additional configuration necessary to ensure PV forecast and live PV production are properly accounted for?

Again, thank you very much for this fantastic project and your support!

lieblinger, 2025-07-05:
After running it overnight, it seems like the situation is gradually improving. For today, the forecast still looks incorrect, but by tomorrow it starts to appear more accurate (please see the attached screenshot).

I’ll keep monitoring it and report back if anything changes.

[image]

ohAnd, 2025-07-06:
Hi @lieblinger ,

I've checked your request/ response, from the data perspective, what goes to EOS, it looks fine...

But EOS response showing a wrong optimizazion result. The solar yield is way above your load, but (as you already mentioned) the SOC is constantly decreasing... makes no sense.

And yes the second overview looks much better but only in the second day...
- First Day has still the wrong decsreasing.
- The phase over the night seems right with the discharge of the battery. 
- The second day looks from frist view as expected.

So, from EOS connect point of view it seems everything right - but EOS makes something wrong.
Maybe some points to check/ investigate:
- do you have made some manual addtional config in EOS local config/ per API ?
  - With the currant EOS connect approach there is no need to configure something in EOS
  - and to have no interferring it could be useful for testing to try a fresh (additonal) installation of EOS, maybe in HA with a second instance of the addon
- based on the shown data: it seems a very huge installation with "bigger" numbers ;-) 
  - -> *assumption* (usually only numbers): EOS optimization was maybe developed with some limits internally, so that this bigger numbers running in some corner cases and the evaluation and leveling out between the sources and consumers need a lot of more loops of optimization for the system, or is more often breaking
  - => has to be supervised and if it's not fitting regularly then the data should presented directly to the EOS project to get some more deeper knowledge
  - for bridging that problem if it is given on EOS side, we can think about a down scaler for the all wattage data to EOS and also then an upscaler for the result inner EOS connect - reduces maybe the accuracy - but I guess negligible
- maybe another try - check what will be given, 
   - if you set the load settings to default
      - this will using a very minimal static load instead of the dynamic projection given by the real household data from load sensor
      - `source: default`
      - so you can see if the wrong result has dependencies to the load
      
    - same or simliar with pv forecast - try to set the `power` value to e.g. 10000 watts insteads of > 80000 something ;-) in your current config
    - also other config should have no direct consequences for the optimization on EOS side ...

Overall, there are some mentionings of 'EOS need sometimes more runs to deliver useful values as a result' and will be improved in future.

Let's give it a try with the ideas above or other users have maybe some addtional ideas...

closed 2025-07-06 by lieblinger