ohAnd/EOS_connect #76: Expense and Income calculation is wrong, battery charging plan doesn't make sense

nanoBit93, 2025-07-22:
Hey,

I've just set up EOS and EOS connect. That is my price config:

`source: default
token: ""
fixed_24h_array: ""
feed_in_price: 0.07
negative_price_switch: false`

These are the current results I'm getting:

[image]

As you can see in the screenshot at 11:00 it wants to charge 5kWh at 5,3ct/kWh... but it shows 0€ Expense and even 21 cents income? Also the values in the other hours don't really make sense to me. 
Probably as a result the charging plan is completely off. Technically I wouldn't need to charge at all, because the load is at about 12kWh / day and the solar forecast ist 25kWh. However, EOS wants still to keep a high battery level for whatever reason. The concerning problem is the times it suggests to charge. Here's the screenshot:

[image]

As you can see for some reason it actively avoids charging in the cheapest hours..

ohAnd, 2025-07-23:
with the shown low load it makes no sense to re-charge the battery so much ... you're right
with given price config you're using the stock price given by akkudoktor api will be used - it's fine or also enough to give eos the chance to decide according to the price

can you please show the rest of your config? (without passwords etc.) to see what are the entries for battery (size, min, max soc...) and so on...

And unfortunately eos needs sometimes some rounds of creating/ re-caculate the model with the given values to show a useful result... means after 2-3 rounds the result could be better and finally useful.

ohAnd, 2025-07-23:
> One additional detail which I just found... the amount of energy stored in the battery at the moment seems to be calculated wrongly. As you can see in my settings I have a 23kWh battery. Current SOC is 74.4%, but EOS shows 14.5kWh as the currently charged capacity, although the real value is about 17 kWh. This could also explain why it still wants to charge that much.
> 
> EDIT: I just noticed the difference in the battery level is related to the min percentage I put in EOS, hence it calculates only with 90% of the total battery capacity, so the numbers match.

Yep, that's right and these limits will also given to EOS.

nanoBit93, 2025-07-24:
For the moment it seems okay again... after the 1st day was over it didn't try to charge from the grid again. Is there any way in the UI to see when it plans to charge the car or the additional load?
And just to confirm. The load_sensor is supposed to include the power that goes into the car and the additional load? If this is the case, my sensor setup is incorrect - my load sensor is already deducted from the cars and additional appliances power.

ohAnd, 2025-07-25:
means after some "more" runs/ cycles the optimization were stabilised and it shows acceptable values?

---

``` yaml
load_sensor: sensor.hausverbrauch_w # => overall household consumption incl. car and additonal
car_charge_load_sensor: sensor.evcc_charger_leistung_w # => pure load of car charging (one or more as overall charging value) it is part of the overall load
additional_load_1_sensor: sensor.evcc_warmepumpe_leistung_w # => addtional load e.g. heatpump - it is also a part of overall load
```

car and addtional load will be substracted from overall load for the prediction - so that both have no influence to the optimization directly
with additional_load_runtime and consumption there is something to play (at this time) to get optimized time for the addtional load - but with the current kind of integration in EOS there needs some ideas how it can be used 'useful'

If you already have e.g. car load deducted - then you get to low load power for that time where car charging was active in the last 2 weeks ... and with that e too optimistic prediction by EOS for the overall needed energy...

nanoBit93, 2025-07-25:
Ah okay that might explain the unexpected behavior. It was fine for 2 days now but tomorrow and the day after it shows the same results again unfortunately:
[image]

1. "hausverbrauch_w"has always been only my sensor for untracked consumption, therefore without car, heatpump and ACs... For now I also removed the car and heatpump sensors from EOS Connect settings.
2. I've created a  new sensor which now collects data over the next two weeks that includes the AC, car and HP and will then replace hausverbrauch_w with it.

Interestingly after doing step 1 the forecast changed completely, even the solar forecast is very different:
[image]

The only difference between the two screenshots is 10 minutes and this config piece:

> (.....)
load_sensor: sensor.hausverbrauch_w
car_charge_load_sensor: "" 
additional_load_1_sensor: ""
additional_load_1_runtime: 0
additional_load_1_consumption: 0

**Finding 1**
The right forecast on screenshot 1 looks like the left one on screenshot 2... and the left one on screenshot 1 looks very familiar as the one I had yesterday... I thought I saw it as well yesterday midnight that the forecast of the previous day appeared but I wasn't sure. I think here's the proof though. It appears that the last forecast run before midnight returns the results of the day it's running instead of the next day? Example:
Screenshot 1 was taken at July 26 00:04. It shows the forecast was received 1 minute before (I've set it to run every 10 minutes), so at 00:03. This screenshot shows the forecast of July 25 on the left. 
Screenshot 2 was taken on July 26 00:13, also received 1 minute before, 00:12. This screenshot shows the correct forecast of July 26 on the left. 

**Finding 2**
Adjusting the sensors didn't change a bit. It's still showing off values so I guess it was not related to my wrong sensor setup. Attaching the request and the result:
[Request.txt](https://github.com/user-attachments/files/21438238/Request.txt)
[Response.txt](https://github.com/user-attachments/files/21438239/Response.txt)

ohAnd, 2025-07-26:
I've checked your request and response - the request looks right
- pv_prognose_wh
- strompreis_euro_pro_wh
- einspeiseverguetung_euro_pro_wh
- gesamtlast
- pv_akku
- inverter
all arrays/ entries as expected and similiar (other levels) to my requests or other that already discussed here

but the reponse by eos, based on this data ... make still no sense
- example - for 12 to 1 pm
  - soc at 100% -_> fine_
  AND 
  - ac_charge requested with 0.875 * 5000 W = 4,375 kW _=> wrong_
  AND
  - discharge not allowed _=> wrong_
  AND
  - load / consumption ~ 0,3 kWh for the most of the next hours _-> fine_
  AND
  - income for that hour with 0,57 € for feed-in of 8,23 kWh _-> fine_
  AND
  - but - Netzbezug_Wh_pro_Stunde shows here a "very" small value - 3.7^-10 Wh --- here it should be based on the requested ac_charge something like 4,3 kWh _=> wrong in a dual point view - should be not charging, and when ac charging it comes from grid and then this values makes no sense..._ 

-> same for the other points where ac charge is demanded (without a need, from my perspective)
=> that leads also to the "wrong" shwoing in the digram with negative grid values (gray columns) - here I expect values by eos that have to be directly addable - PV power [orange] + AC_charge_power [red] + (Grid - AC_charge_power)[gray]

I can only guessing ... EOS runs something wrong on this kind of data... 

Interesting, but not explainable at this point from my side - or I have a general error in thinking here

maybe you can give a try with akkudoktor (default) as a pv forecast source - only to see if it runs in the same behaviour

---
and to have a look also in this direction, which version (there is no versioning ;-) - which commit or something like this) you are running of eos

ohAnd, 2025-07-26:
Ah...
Maybe... 🤔😅

You using prices, it seems directly from epex without any taxes and fees... Means not tibber or the other ones.
(can you share your config here also)

You get in that false phases lower prices for grid input than you can earn with feed in... You have configured  here it seems 7ct / kWh.

So maybe EOS optimizing here in this way...  the needed local energy will be bought and all own produced will be sold.

Maybe you can try at first with 0 feedin and than maybe with another price source or feedin price without taxes and so on (means all prices have to be in the same base)

nanoBit93, 2025-07-27:
yep, I guess you found the issue! The problem is that the config doesn't allow me to add surcharges on the price according to the config docs. This is my config:

> source: default
token: ""
fixed_24h_array: ""
feed_in_price: 0 #-----> before, I had 0.07 here
negative_price_switch: false

This is how it looks:
[image]

I also switched back to openmeteo for the forecasting which changes the plan to this:
[image]

ohAnd, 2025-07-27:
nice ...
So, if you don't use tibber or smartenergy - maybe you can tell which one you use ans maybe there is a possibility (api) to add this source, too.
In the meantime please adjust your feedin price then to the price without tax and fees - so it have the same base as the stock prices - to get the right weightening for EOS.

Nevertheless, I will extend the description for prices - to have this dependency directly visible.

ohAnd, 2025-07-30:
Ok, yep, there is nothing... ;-)

For the base optimization the overlays of taxes and fees are not relevant, if all used price are on the same base as discussed.

The use of the real prices is only relevant for the displaying of expense and income.

If there is a need we can think about to configure additonally the parts for fix and dynamic costs - see for that the longer discussion here #51 
Maybe we can discuss then in a new issue - and we can close here

nanoBit93, 2025-08-01:
I agree, I posted my answer in #51 . Let's close here. Thanks for your super quick help!

commit 81fd5d1b (2025-08-03): docs: update electricity price configuration with important notes on tax/fee basis - fix #76