ohAnd/EOS_connect #78: Load Sensor Fehlende Daten

cookie2512, 2025-08-02:
Hallo zusammen,

ich bekomme in den Logs immer wieder das Problem, dass Daten fehlen für den Load Sensor.

Hier der Auszug aus dem Log:

`2025-07-30 09:21:48 INFO [EOS] OPTIMIZE request optimization with: http://172.18.0.1:8503/optimize?start_hour=9 - and with timeout: 180
2025-07-30 09:22:33 INFO [EOS] OPTIMIZE response retrieved successfully in 0 min 45.56 sec
2025-07-30 09:22:33 INFO [Main] Overall state not changed recently - remaining in current state: MODE DISCHARGE ALLOWED  (_____OOOOO_____)
2025-07-30 09:22:33 INFO [Main] Next optimization at 09:25:33. Sleeping for 2 min 59 seconds
2025-07-30 09:25:33 INFO [Main] start new run
2025-07-30 09:25:33 INFO [PRICE-IF] Prices updated
2025-07-30 09:25:33 INFO [LOAD-IF] creating load profile for weekdays 2025-07-23 00:00:00+02:00 (Wednesday) and 2025-07-24 00:00:00+02:00 (Thursday)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 00:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 01:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 02:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 03:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 04:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 05:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 06:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 07:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 08:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 09:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 10:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 11:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 12:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 13:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 14:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 15:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 16:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 17:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 18:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 19:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 20:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 21:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 22:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:36 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-16 23:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:38 ERROR [LOAD-IF] Error processing energy ('sensor.ac_loads') data at index 3637: {'state': '1329.1184', 'last_updated': '2025-07-24T13:55:26.596590+00:00'} (next: {'state': 'unknown', 'last_updated': '2025-07-24T13:55:51.283425+00:00'}) - could not convert string to float: 'unknown' (check: http://172.18.0.1:8123/history?entity_id=sensor.ac_loads&start_date=2025-07-24T11%3A55%3A26.596590%2B00%3A00&end_date=2025-07-24T15%3A55%3A26.596590%2B00%3A00)
2025-07-30 09:25:38 ERROR [LOAD-IF] Error processing energy ('sensor.ac_loads') data at index 3638: {'state': 'unknown', 'last_updated': '2025-07-24T13:55:51.283425+00:00'} (next: {'state': '1341.2231', 'last_updated': '2025-07-24T13:56:14.013789+00:00'}) - could not convert string to float: 'unknown' (check: http://172.18.0.1:8123/history?entity_id=sensor.ac_loads&start_date=2025-07-24T11%3A55%3A51.283425%2B00%3A00&end_date=2025-07-24T15%3A55%3A51.283425%2B00%3A00)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 00:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 01:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 02:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 03:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 04:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 05:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 06:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 07:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 08:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 09:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 10:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 11:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 12:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 13:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 14:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 15:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 16:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 17:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 18:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 19:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 20:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
2025-07-30 09:25:39 WARNING [LOAD-IF] load = 0 ... Energy for 2025-07-17 21:00:00+02:00:   0.0 Wh (sum add energy   0.0 Wh - car load:   0.0 Wh)
[... cut ...]
Ich habe vor einigen Tagen erst erkannt, dass ich meinen AC-Loads Sensor in Home Asisstant falsch erstellt hatte, da dieser negative Werte aufzeigte, bedingt dadurch dass der PV-Erzeugungssensor träger war als der Sensor am Netzübergabepunkt. Außerdem hatte ich vergessen den Akku einzubinden.

Außerdem sehen die Werte nicht richtig aus im Graphen. Betrifft vor allem den Load Sensor und die Prognose des Akku SOC. (Siehe Screenshot)
Ich denke das liegt aber wie oben beschrieben, dass der Load Sensor negative werte hatte, oder?

[image]

ohAnd, 2025-08-02:
Hi @cookie2512 ,

zum ersten Punkt - Daten aus der Vergangenheit nicht mehr in der Historie sind:
=> https://github.com/ohAnd/EOS_connect?tab=readme-ov-file#getting-historical-values
Bitte mal den Punkt checken!

Load-Sensor Werte können positiv oder negativ sein - diese werden in der Berechnung entsprechend immer als positive Werte weiterverarbeitet. (werde ich in der config readme noch entsprechend nachziehen)

Zu deinem Screenshot - die Load-Values (grüne Balken) sind in etwa den AVGs aus deinen HA Screenshots ... 
(AVGs = stündliche Mittelwerte aus dem historischen Verbrauch, der nun ja nur den einen Tag vor 7 Tagen auswertet, da die älteren noch nicht persistiert sind)
daher bei den eingehenden PV Leistungen sieht für mich erstmal auch der SOC Verlauf valide aus... an den Income Werte sieht man ja auch die entsprechend hohe Einspeisung. 

Also mal die retention hoch setzen und dann die Ergebisse vergleichen ggfs. sonst nochmal ein DEBUG log mit den gelesenen load Werten posten (oben ist ja nur der Fehleranteil zu sehen)

cookie2512, 2025-08-02:
Unterm Tage passen die Load-Werte, aber die werte so ab ca. 20 uhr bis früh um 6 uhr passen nicht. aktuell bin ich früh um 6 uhr meist bei 40% akku stand.

Hier mal der Sensor für die letzten Tage nachts:

[image]
[image]

Hier Akku SOC:

[image]
[image]

Hier was EOS prognostiziert für heute nacht:

21: 0,080
22: 0,074
23: 0,065
00: 0,065
01: 0,065
02: 0,065
03: 0,065
04: 0,065
05: 0,071

Hier ein Screenshot von HA Energie heute Nacht:

[image]

Mir kommt es vor als würde er das um eine Nachkommastelle falsch berechnen.

ohAnd, 2025-08-02:
Wie hier beschrieben https://github.com/ohAnd/EOS_connect?tab=readme-ov-file#collecting-data werden die Daten aus der Historie von heute und morgen minus 7 Tage und minus 14 Tage jeweils stündlich gemittelt genutzt. (die minus 14 sind dann erst mit retention und erfolgter Datensammlung verfügbar)

Wenn ich den ersten Screenshot aus Deinem ersten Post ansehe - scheint zumindest die Nachtphase hier auch kleiner als die der letzten Tage ... 
Darum schau mal bitte genau in den Zeitraum von vor 7 Tagen (so wie in Deinem letzten Post nur eben genau in diesen Zeitraum) und vergleiche dann die Daten die in EOS connect angezeigt werden.

Der Versatz den du meinst muss hier daran liegen (und damit zufälligerweise), da der Wert ja in Watt richtig vorliegt. Die Tageswerte aus dem Dashboard Screenshot sind ja überschlagsweise entsprechend dem was in der Historie bei Dir im Bereich 25.7. (Fr) / 26.7. (Sa)/ 27.7. (So) zu sehen ist...

Und so wäre das Optimierungsergebnis auch valide bzgl. des sehr schwachen SOC Abfalls über Nacht, wenn hier nur ~ 70 Wh pro Stunde abgenommen wird...