ohAnd/EOS_connect #80: How to determine whether EOS_connect/EOS are running?

HolBaum5, 2025-08-05:
What is the best way to determine whether EOS_connect and EOS are running? 
If neither of them are working, I would like to switch to normal mode on the inverter.

ohAnd, 2025-08-06:
Hi @HolBaum5 ,

for EOS connect you can check if `http://<your-ip>:8081/json/current_controls.json` is reachable and for double check if there is in the returned json `timestamp` with a current value.
```json
{
    "current_states": {
...
    },
    "evcc": {
...
    },
    "battery": {
...
    },
    "inverter": {
...
    },
    "state": {
...
    },
    "eos_connect_version": "0.1.22",
    "timestamp": "2025-08-06T22:00:39.881865+02:00",
    "api_version": "0.0.1"
}
```

For EOS there is an API endpoint for health: `http://<your-ip>:8503/v1/health` and returns
```json
{"status":"alive","pid":1}
```

btw. for the last month's I'm not see any outages of EOS_connect and EOS within my Homeassistant Enivronment

HolBaum5, 2025-08-09:
@ohAnd Thanks for the tips, I'll build my program using the information.