ohAnd/EOS_connect #93: No control data in optimized response

maseb24, 2025-09-13:
Hallo,

ich habe EOS Connect als Add On in Homeassistant laufen.

Von EOS bekomme ich immer wieder die Fehlermeldung

2025-09-13 18:16:59 INFO [EOS] OPTIMIZE request optimization with: http://192.168.115.55:8503/optimize?start_hour=18 - and with timeout: 180
2025-09-13 18:17:02 INFO [EOS] OPTIMIZE response retrieved successfully in 0 min 3.29 sec
2025-09-13 18:17:02 ERROR [EOS] OPTIMIZE Request failed: 400 Client Error: Bad Request for url: http://192.168.115.55:8503/optimize?start_hour=18 - response: <Response [400]>
2025-09-13 18:17:02 ERROR [EOS] RESPONSE No control data in optimized response
2025-09-13 18:17:02 INFO [Main] Next optimization at 18:47:02. Sleeping for 29 min 59 seconds

Woran kann das liegen?
Für eine Hilfe bin ich sehr dankbar

Gruss Rudi

ohAnd, 2025-09-13:
Hallo Rudi,

Bitte mal loglevel in der config auf debug setzen. Dann wird der request und response zum/ vom EOS im Detail angezeigt.

Damit lässt sich dann mehr sagen.

maseb24, 2025-09-14:
Hallo,

anbei das Protokoll bei loglevel debug

Gruss Rudi

[855a9a00_eos_connect_develop_2025-09-14T16-09-20.026Z.log](https://github.com/user-attachments/files/22321796/855a9a00_eos_connect_develop_2025-09-14T16-09-20.026Z.log)

maseb24, 2025-09-17:
Hallo,

ich habe die optimization hours und die prediction hours auf 48 Stunden gestellt.

Der Fehler ist weg. EOS connect liefert nun ein Ergebnis.

Gruss Rudi

Vielen Dank

commit c4b38aad (2025-09-18): feat: add EOS configuration requirements section with prediction settings and troubleshooting guidance - see No control data in optimized response
Fixes #93