ohAnd/EOS_connect #98: smartenergy_at prices deviate

Joebinator, 2025-09-29:
Hi there!

Somehow i am not sure what prices are being pulled when i use "smartenergy_at"

A quick research at awattar for example 2025-09-29 at 10:00 gives me the prices:

Germany:  0,09021 €/kWh (without charges, taxes, etc)
Austria: 0,0926 €/kWh (without charges, taxes, etc)

EOS Connect shows me:  0,111 €/kWh

[image]

Did i configure something wrong?

`price:
  source: smartenergy_at
  token: ""
  fixed_24h_array: ""
  feed_in_price: 0.0744
  negative_price_switch: false`

ohAnd, 2025-10-01:
Hi,

these prices are given by the smartenergy.at API see:
=> https://apis.smartenergy.at/market/v1/price --- [see also here](https://github.com/ohAnd/EOS_connect/blob/23d4ea72af089b88915b90929391b78616d91828/src/interfaces/price_interface.py#L42C1-L42C64)

based on the doc at https://www.smartenergy.at/api-schnittstellen

maybe the difference you discovered is this:
```
Spezifikation: |  
tariff | Tarif, string
unit | Einheit des Werts von data:value, string
interval | Gültigkeit des Preises in Minuten, integer
data:date | Preis gültig ab, lokales Datum und Uhrzeit
data:value | **Preis inkl. 20% MwSt.**, dezimal

```

ohAnd, 2025-10-02:
got it - thanks for detailed investigation - was primitive rounding problem

commit 0361136d (2025-10-02): fix: increase precision of hourly price calculations to 9 decimal places - fix smartenergy_at prices deviate
Fixes #98
closed 2025-10-18 by ohAnd